heliorc/imuf-release #4: acc calibration broken

adrianmiriuta, 2018-04-13:
no level flight possible, quad drifts !

```
# diff

# version
# ButterFlight / HELIOSPRING (HESP) 3.5.1 Apr 12 2018 / 19:25:14 (eee3a0ecb) MSP API: 1.40
# IMU-F Version: 106

# name
name Warp-271

# resources

# mixer

# feature
feature -LED_STRIP
feature -OSD
feature ESC_SENSOR
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map
map RETA1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2075 0
aux 1 1 0 1525 1650 0
aux 2 2 0 1700 1825 0
aux 3 6 2 1950 2075 0
aux 4 7 3 1950 2075 0
aux 5 13 1 1425 1550 0
aux 6 26 1 1950 2075 0
aux 7 30 2 1450 1550 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 1
set imuf_rate = 32K
set acc_calibration = -37,53,-85
set min_check = 1000
set max_check = 2000
set serialrx_provider = FPORT
set serialrx_inverted = ON
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1500
set vbat_warning_cell_voltage = 34
set current_meter = ESC
set battery_meter = ESC
set small_angle = 180
set deadband = 7
set yaw_deadband = 11
set pid_process_denom = 2
set frsky_unit = METRIC
set debug_mode = NOTCH

# profile
profile 0

set dterm_lowpass_type = PT1
set p_pitch = 61
set i_pitch = 47
set d_pitch = 23
set p_roll = 61
set i_roll = 47
set d_roll = 23
set p_yaw = 61
set i_yaw = 47
set d_yaw = 23
set d_level = 50
set horizon_tilt_effect = 50

# rateprofile
rateprofile 0

set thr_mid = 13
set thr_expo = 67
set tpa_rate = 70

#
``` 
quad is on level plane !
calibration was performed !

[image]

iLuvKwdzMan, 2018-04-14:
having issue with board orientation, it constantly shows its upside down even changing in config doesn't seem to change anything,  i cant get it off the ground from FW ver. RC3 and up , none of the new ones work, i have to revert back to 3.4.3 just to get it flying.  i dont know where to find more info,

iLuvKwdzMan, 2018-04-14:
i also seem to be having gyro drift , in acro mode, it drifts to the left, trims all set at 1500 in BF config but still does this

jasonsmit4, 2018-04-15:
I'm having the same issue as iLuvKwdzMan.  Anything beyond RC3 is an unflyable tasmanian devil for me.  "align_board_pitch = 180" seems to have no effect in my case.   Acc calibration in the configurator also seems to have no effect whatsoever.

adrianmiriuta, 2018-04-16:
RC8 acc calibration broken (still)

closed 2018-06-14 by orneryd