heliorc/imuf-release #12: Flashing betaflight hex, hangs the Helio in DFU mode

emski0, 2018-07-10:
Hi
I am trying to flash helio with the supplied betaflight hex file in this repo, but the helio hangs in DFU mode and FLASHING (goes immediately to 100% after erase) and hangs in there.
Is the betaflight hex file not ready yet? or what's the solution?
Thanks

toper74, 2018-07-10:
Hi,
Which version ? Did you try to download the hex one more time (or try the zip) ?
Try to post on [#Slack](https://flyhelio.slack.com/messages/CASH64HSM).

emski0, 2018-07-10:
I tried 3.4-RC4 and 3.4

orneryd, 2018-07-10:
check your cable, I just re-verified flashing the .hex file and it finishes and reboots successfully.

closed 2018-07-10 by orneryd

emski0, 2018-07-11:
Using the zip file and unzipping worked. However directly using the provided hex hangs on flashing in DFU